kosme/arduinoFFT #1: Example F02

Pranavgulati, 2016-12-30:
hi 
i was trying to use the library but i couldnt understand what the output was for 
i want to use the ADC on the esp8266 to take the FFT of the incoming signal on the ADC values, is this the correct library to use and if yes where should i provide the values from the ADC.

i felt like the amplitude variable could store the values incoming from the ADC but i'm not so sure, any help is appreciated :-)

kosme, 2016-12-30:
First of all, thank you for using this library. 

What Example F02 does is: iterate through a range of frequencies, generate a "perfect" sinusoidal signal with each of those frequencies, and print the frequency at which it generated the signal, the frequency the FFT library obtains from the signal, and the time it took to do the FFT calculation. 

Indeed you can use the library for what you want, check example 1 as it first generates a "perfect" sinusoidal signal and then analyzes it. You should define the number of samples you want to use for the calculation and you should store the values from the ADC on the equivalent of vReal[samples]

Your code should look something like this:

`#include "arduinoFFT.h"

arduinoFFT FFT = arduinoFFT(); /* Create FFT object */

const uint16_t samples = 64; //This value MUST ALWAYS be a power of 2
const double samplingFrequency = 5000;

double vReal[samples]; 
double vImag[samples];

int i = 0;

void setup()
{
  Serial.begin(115200);
  Serial.println("Ready");
}

void loop() 
{
  /*Store values from ADC in vReal until it is filled according to samplingFrequency*/

  FFT.Windowing(vReal, samples, FFT_WIN_TYP_HAMMING, FFT_FORWARD);	/* Weigh data */
  FFT.Compute(vReal, vImag, samples, FFT_FORWARD); /* Compute FFT */
  FFT.ComplexToMagnitude(vReal, vImag, samples); /* Compute magnitudes */
  double x = FFT.MajorPeak(vReal, samples, samplingFrequency);
  Serial.println(x, 6);
}`

Pranavgulati, 2016-12-31:
thanks a lot for the reply,
i think i get it now the vReal array is where the incoming signal gets first stored before processing with corresponding vImag set to 0. After the compute is done the result is also stored in vReal and vImag but after the complexToMagnitude is called the magnitude is then stored in vReal
(leaving it here as an explanation)

Thanks for your help, i'll test it asap, closing the  issue :-)

Happy new year !!

closed 2016-12-31 by Pranavgulati