kosme/arduinoFFT #2: Does it work on ESP8266？

Fireflaker, 2017-04-13:
ESP8266 has a much faster ADC and should be able to this job much better. Will it run properly on ESP8266 using Arduino compiler?

kosme, 2017-04-14:
Should work smoothly but still I will give a try on these days and report
back.

kosme, 2017-05-03:
Just checked. I tried both examples. It does work smoothly. Just be careful when using the ADC, some ESP8266 boards have a limited range of voltage they can handle on the ADC (Ex. Sparkfun Thing Board)

closed 2017-05-03 by kosme

brendanmatkin, 2017-05-03:
I think all ESP8266 are only rated for 0.0-1.0v input on the ADC. Good idea just to put a simple voltage divider on the input. I think some breakouts (like the wemos d1 and NodeMCU) already have that voltage divider in place.