kosme/arduinoFFT #3: samples could not more than 128 ?

marvinusarif, 2017-06-02:
Hi, 

I would like to ask about the number of samples - why the number of samples can't be greater than 128 e.g. 256, 512. I am using redbear duo / particle spark and it does not return error - however, it is not working and the system were halted.  Looking for your help. thanks..

kosme, 2017-06-02:
I'll check it out

VojislavM, 2017-08-01:
I have the same problem...

kosme, 2017-08-04:
Just fixed it on release v1.2.

closed 2017-08-04 by kosme

kieferlord, 2020-03-14:
Hello. Your library works great for sample sizes of 128 and below, but when I try anything above that in order to obtain a better resolution, I have storage issues. The specific error message is:

Global variables use 2278 bytes (111%) of dynamic memory, leaving -230 bytes for local variables. Maximum is 2048 bytes.

I have tried moving the vReal and vImag arrays into the void loop() so they would no longer be considered global variables, but this caused a bunch of other problems (crazy things printing, like backwards question marks, etc.). Do you have any suggestions? Thanks.