kosme/arduinoFFT #4: Your FFTcode  just  run  only one time??

Feng-Ching, 2017-07-04:
Hi,

I'm so sorry my English is bad,and
May I ask you some question about your arduino code??
Your arduinoFFTcode just  run  only one time??
I saw the data ,the first data is yes,but  The value will accumulate again~again,until overflow
if I wanted to modified this code to do compute FFT again~again,and it couldn't  accumulated 
how can i do ><

kosme, 2017-07-05:
Are you running any example in particular or is it a custom code you
made? Which
value is accumulating or not accumulating? (I could not understand if you
wanted it to accumulate or not)

Feng-Ching, 2017-07-05:
[image: 內置圖片 1]
Thanks ,your message , I use arduino mega2560 and didn't  give A0 signal
 ,and just modified " line 80 " to  " //while(1) ".and ... the first value
looks normal

[image: 內置圖片 2]
but...the last the value will be ovf .. inf .. nan ...
[image: 內置圖片 3]
 I hope FFT can always run ,and the value couldn't  accumulate .

there is a test I gave  A0 sinewave and i take value to Excel compare
Looking for your  Professional help!!

closed 2017-07-11 by Feng-Ching

Feng-Ching, 2017-07-11:
I think I know where my problem is.
Thank you for your willingness to help
and the selfless offer of the library!!

All my best blessing to you!👍

2017-07-05 21:10 GMT+08:00 張封卿 <a0935022728@gmail.com>:

kosme, 2017-07-11:
I think it would be useful for other users if you share what you learned
and where your problem was.

> [image: 內置圖片 1]
> Thanks ,your message , I use arduino mega2560 and didn't give A0 signal
> ,and just modified " line 80 " to " //while(1) ".and ... the first value
> looks normal
>
> [image: 內置圖片 2]
> but...the last the value will be ovf .. inf .. nan ...
> [image: 內置圖片 3]
> I hope FFT can always run ,and the value couldn't accumulate .
>
> there is a test I gave A0 sinewave and i take value to Excel compare
> Looking for your Professional help!!
>
>
>> Are you running any example in particular or is it a custom code you
>> made? Which
>> value is accumulating or not accumulating? (I could not understand if you
>> wanted it to accumulate or not)
OAO855R8_0Zks5sK4HBgaJpZM4ONi41>

—

<https://github.com/kosme/arduinoFFT/issues/4#issuecomment-314481267>, or mute
the thread