kosme/arduinoFFT #9: Memory limit exceeds for samples = 256 in Arduino Uno

ashish28ranjan, 2017-10-22:
When using samples = 256, the Global Variables use more than 2048 bytes which is not allowed in Arduino Uno (ATmega328p).
I want to calculate DFT of only real signals (e.g. ADC input). Could it be possible to optimize the code for only real signals?
I tried to do it myself but came to the conclusion that the code was already immensely optimized for space.

kosme, 2017-11-30:
Yeah, i know of that problem. (256 samples not fitting into Arduino RAM) Unfortunately, FFT depends on the imaginary part for doing the math. Maybe you could try looking for a FHT algorithm, that one only uses the real part of the signals. Another option is changing the hardware for something with more RAM. I have seen the code working on esp8266 and pic32 boards.

closed 2017-11-30 by kosme

ashish28ranjan, 2017-12-01:
Thanks bro! I would look into it.

akshulg, 2018-02-16:
128 number of samples not enough to calculate fft of real signals?

kosme, 2018-02-16:
Mostly it depends on how stable is the signal you are measuring. If it is
stable enough, 128 samples is enough if you choose the correct sampling
frequency (around 2.5 times your maximum expected frequency) Having a
really high number of samples and/or sampling at a really high frequency will
not provide an important increase on accuracy as the relationship is
logarithmic. Also, increasing the number of samples increases the
processing time.