kosme/arduinoFFT #10: Compatibility with ARM Mbed Compiler?

architm12, 2017-11-30:
Hello,

I intend to use this library with the FRDM K64F on the ARM Mbed Online Compiler. I wanted to inquire whether I can use this library as it is, or will I have to make any changes to the existing library.

Thank you!

kosme, 2017-11-30:
Just checked on the mbed platform. You have to do minor modifications. Just removing and adding comments. Follow this instructions.

On file arduinoFFT.h comment out line 33 #include <avr/io.h>

On file types.h uncomment lines 24,25,26,27

That will do the trick. I will do the proper changes to the repo and push the changes soon.

closed 2017-11-30 by kosme

architm12, 2017-11-30:
I really appreciate your speedy help. 
Looking forward to using your mbed repo!
Thank you

kosme, 2017-12-05:
I see the problem. The library is not hosted there and they are not the
same, even if the video guy says it is. You can get here what you are
looking for. https://docs.google.com/open?id=0B9XPtn5QO-OzWF9ZQnVIcDV6THc

architm12, 2017-12-05:
Also, I tried using your library with the suggestions you mentioned, but still I wasn't able to use the library to my use on the ARM Mbed Compiler. If you could publish your mbed repo soon, it would be really great!
Thank you

kosme, 2017-12-05:
I am no quite sure it didn't work, I tested i first on the online compiler and Platformio. Anyways, I will not do a separate repo for mbed. Probably tonight I will push the changes in code for it to work also on mbed platforms