kosme/arduinoFFT #12: Doubt

akshulg, 2018-02-16:
@kosme Not able to understand what are the meaning of the 2 plots, red and blue, in the serial plotter. Using FFT_03 because I am providing external input from Sparkfun sound detector. Please help!

akshulg, 2018-02-16:
@kosme what is the use of PrintVector function in FFT_03 code? you didnt mention that. Please tell me about that function.

kosme, 2018-02-16:
For your first doubt, combine the sampling code of example 03 with the
printing code from example 04.

PrintVector is an auxiliary function to show the contents of the array. It
is not properly part of the fft process.

akshulg, 2018-02-16:
Thanks for the help. I got it. my need is just to print the computed magnitudes for all frequencies. I commented other print statements of weighted data, real and img values. Its working.

akshulg, 2018-02-17:
one more doubt. the data used in fft is 2N(real+imaginary). Calculating memory required for say 256 samples it would be 2*256*1byte(each) = 512 bytes (arduino is 8-bit). And memory in arduino is 2048 bytes.
So why not able to use 256 samples?

kosme, 2018-02-17:
It uses float data type. According to the compiler documentation, each
float is 4 bytes long. So 256 x 2 x 4 = 2048 bytes only for storing the
data. And it still needs memory space for another bunch of things so it
would never fit.

akshulg, 2018-02-17:
Thanks once again!

closed 2018-02-17 by akshulg