kosme/arduinoFFT #13: dBA values

akshulg, 2018-02-28:
So, actually I am using this library to get sound pressure level dBA values in frequency domain.
Is the weighted data calculated in the FFT_03 example a-weighted data so that the final computed values are in dBA itself?

kosme, 2018-03-03:
No, the weighted data in any of the examples is the result of the windowing function applied to the samples. dBa values are computed based on the correlation between the "intensity" of the signal and the frequency of that signal. The windowing functions computation is based on the position of the sample within the samples array.

noisecontrol, 2018-05-26:
Hello 
How can i get adc in octav band freqency
(16-32-63-125-250-500-1000-2000-4000-8000-16000Hz)
I wany to change adc to dB and then convert to dBA