kosme/arduinoFFT #14: Can one do FFT only for some of the spectrum?

BogusException, 2018-04-28:
@kosme I appreciate your work. I have a question. 
How can I only sample a _portion_ of the frequencies? If I am only interested in 500Hz to 1kHz (example), can I then have faster processing, and smaller bin width? I think it is too much processing to do the whole spectrum in my project.
Can FFT be made faster, more fine-grained, if it only looks at a subset of the whole? If so, any ideas on how?
Thanks!

kosme, 2018-05-16:
I don't think you can do that . As far as I know, that is not how the Fourier Transformation works.

t3db0t, 2019-04-04:
I found this [interesting article](http://www.fftw.org/pruned.html) on "pruned" FFTs, but it seems to indicate that performance gains are not necessarily that great.