kosme/arduinoFFT #15: look up table

laurentopia, 2018-04-29:
Could c1 and c2 be precalculated to avoid sqrt?

laurentopia, 2018-05-17:
oh no... i wonder if the beta 1.9 of the ide contains a fix to the compiler

kosme, 2018-05-17:
I'll give it a try but wouldn't count on it working. The Arduino guys have been stuck with gcc 4.9 for years

laurentopia, 2018-05-18:
Good to know there is a certain inertia.

kosme, 2019-05-14:
@laurentopia Just tried this again. It is now possible to create a constant table with the values of c1 and c2. The usage of RAM increases and the code footprint decreases a little. But to my dismay, there is not much increase in the code's efficiency in terms of execution time or result's precision. According to some rough testing I did last night, the execution time for example code FFT_01 is reduced at best from 30 to 29 ms. Proportionally that's a reduction of 4% but in terms of the running time, it is almost negligible. Tell me if you are still interested in that feature so I then refine and upload the code.

laurentopia, 2019-05-15:
Thanks for testing this out.
I moved to a MAIX which has hardware FFT so I'll leave it up to you ;)

HorstBaerbel, 2020-02-19:
See my pull request #42 for possible speed improvements.

closed 2024-04-24 by kosme