kosme/arduinoFFT #20: Usual frequencys

guifanti, 2018-08-11:
Hello,
How can i get adc in octav band freqency?
(16-32-63-125-250-500-1000-2000-4000-8000-16000Hz)

emmyprinso, 2019-08-11:
am using ESP8266 I notice the highest frequency one can get is 5KHz is there a fix to that?

kosme, 2019-08-12:
@emmyprinso The maximum accurate frequency is limited by the sampling frequency and the number of samples you take. The sampling frequency should be at least two times the maximum frequency you expect to be present in your signal. The sample size is also important because at high frequencies, a small sample size is not enough information to do accurate calculations. Check those two things

emmyprinso, 2019-08-20:
@kosme thanks you for your response.
Do you know the highest frequency I can get with NodeMCU ESP8266?

nakanaela, 2019-09-06:
If you look at the comment from pratik at https://bbs.espressif.com/viewtopic.php?t=2440, the fast_read api returns results in ~5μs (200kHz) so the highest frequency based on ADC read speeds would be 100kHz. But if you want to go that high, you will need a considerable amount of RAM to hold your samples and you would need a microphone capable of 'hearing' that frequency.

If you are only able to get 5kHz, check the timing of your audio sampler and check the frequency response range of your mic. I sample at 32768 Hz on an ESP32 and depending on the mic, there is a noticeable change in either the high-end or low-end response...all because of the mics. I also collect 2048 samples for processing and use a hardware based timer to ensure that a sample is collected every 1000000/32768μs. reading the ADC takes <15μs, or about half of the time I have available to collect samples at the specified sampling frequency.