kosme/arduinoFFT #28: Possible additional method

vtomanov, 2019-01-02:
Also a method void MaxPeak(uint16_t start_idx, double *f, double *v) is very useful - e.g.  when you want to get the max peak not the major peak - and also filter the first n bins e.g. 👍 

.....
 uint16_t IndexOfMaxY = 0;
      for (uint16_t i = start_idx; i < ((this->_samples >> 1) + 1); i++) {
          if (this->_vReal[i] > maxY) {
            maxY = this->_vReal[i];
            IndexOfMaxY = i;
          }
      }

kosme, 2019-01-03:
I think I am not understanding what you are asking for because I don't see in what situation that method would be useful. Most of the times, the value of the major peak falls between two measurements in the array. With the method you are asking for, that value would be completely ignored

vtomanov, 2019-01-04:
Agree closing the issue

closed 2019-01-04 by vtomanov