kosme/arduinoFFT #32: Request to keep older features to reduce memory footprint

rdragonrydr, 2019-07-14:
As someone recently using this library, I have noticed that the methods supporting pointers to the data to be processed are deprecated, and possibly soon to be removed. I ask that this feature be kept, as passing the pointers to the data in the constructor makes it impossible to allocate and free the rather large memory usage of the library when it's not in use.

Perhaps this could be simplified by making the simplified version of the methods call the existing parameterized functions with the _internal_ pointers and values as arguments?

At minimum, the ability to deallocate the imaginary portion of the data (since most results are returned in the real portion of the arrays) would be helpful, but I have personally found it more efficient to copy the results to a smaller type and free both arrays.

One more thing I'd like would be a filter to normalize all the data to a certain range (so outputs cannot go higher than 100, for example. I'm not sure how this could work since the volume is unknown and I don't see the highest result **always** being 100 as practical), and perhaps apply a minimum amplitude to the results below which the bins are zero so that visualizers (for instance) don't have spurious noise. I'm not certain if this is already a feature...?

HorstBaerbel, 2020-02-21:
> As someone recently using this library, I have noticed that the methods supporting pointers to the data to be processed are deprecated, and possibly soon to be removed. I ask that this feature be [...]
Could you describe a use case or post code snippet showing what you want to do?  
If having only a constructor, would a function `void setData(T *vReal, T *vImag);` (set new pointers. pick a name) or `void swapRealImag();` (swap internal pointers. pick a name) to enable users to switch / swap / replace data pointers however they like suffice for you.

rdragonrydr, 2020-03-06:
Okay, that's useful. I was also recently considering changing the datatype, too, so that's really useful.

Also, my use case was for running a FFT on an ESP8266, so I had to load up both arrays and process them, but I only need a scaled-down magnitude of the 'real' part (for a spectrograph, basically). Since it's run in a loop, and I have a few other pieces of code using an astonishing amount of memory (bunch of background JSON parsing and a web server), I wanted to deallocate the imaginary (at minimum--I actually made a byte array that stores the scaled values and deleted the real one too) array when it wasn't actively doing the transform. But with the constructor setup this used, I had problems adding it to an object and I couldn't delete the memory when it was done (since it was only set in the constructor). So, this has solved my problem.

Funny aside: I later realized that I could probably have also solved this by dynamically allocating the FFT object too, and just instantiating all of them when needed...

closed 2020-03-06 by rdragonrydr