kosme/arduinoFFT #33: Finding real amplitude from computed magnitude

skeletozaure, 2019-09-11:
Hello Enrique,

I'm not a math expert, i'm just trying to use as best as i can your neat library. i have seen in the examples that computed magnitude in the Freq/Magnitude is not the real amplitude of the signal in the Time/Amplitude domain. I have seen here and there that to find the real amplitude from the magnitude with a FFT at a given Frequency, it's possible to divide by N or Sqrt(N). N is the number of samples used to compute the FFT. By the way, even if it works sometimes, it doesn't work everytime.

For example, Actually, i'm working on a project with an ESP32 and an ADXL337 (accelerometer), which purpose is to detect vibration frequency and amplitude on mechanical devices.

ADXL337 sensitivity is -3G/3G on 3 axis, with a linear response.  From what i have seen, max sampling rate is 1600Hz on X&Y and 550Hz on Z. which let me work in the frequency range 0->800Hz on X&Y and 0-275Hz for Z. 

I'm using only X axis for testing. I'm sampling at 512Hz and i'm using 1024 samples (which means that my acquisition lasts 2s, but no problem, because i don't need real time).

Shaking the ADXL337 by hand, i can actually achieve easily 5/6 hz Freq. The result of the FFT is nice for the frequency, which gives me a huge peak at 5.5 hz.

Here are the results graphed in a Google sheet for better understanding :

[https://docs.google.com/spreadsheets/d/1apywfbcAPRnyqvjKSxycdRW7zb4gMOXLaoHnKnQRkIs/edit?usp=sharing](url)

Someone can say that i have checked the ADXL337 to strongly, I went beyond the limits of the accelerometer. But i don't think it has a big impact on magnitude (maybe i'm wrong also...).

On the ESP32 the range of the ADC is 0-4095, with no negative value (of course). Using ArduinoFFT  i have an amplitude of 613881 at 5.5Hz

I'm wondering, how can i find what was the amplitude in the Time/Amplitude domain at 5.5hz from this value. it's probably around 4095. Is there some mathematical relationship or should i find this relationship with some calibration samples ?

Thx for your help
Cédric

kosme, 2019-09-19:
Several things. I saw your graphs and at some points, there is indeed some saturation on the amplitude that shows as flat regions instead of defined, sharp peaks. That indicates that the acceleration the sensor is experiencing is beyond the +-3G that it can measure. But that phenomenon in that quantity should have a negligible effect on the frequency calculation. 

I still have not been able to establish a connection between the amplitude of the physical signal and the magnitude expressed on the FFT peaks. As I told you on the email you wrote to me, the easiest way right now to measure the movement amplitude is to use the raw data the accelerometer is producing. That will produce the peak to peak amplitude in G's.

JJLynnXiao, 2020-12-01:
Hi, 
I think it probably needs an amplitude correction factor (divide by windowSum/2) for FFT amplitude.
 I add in ComplexToMagnitude function and amplitude is consistent with my test signals.
Hope this helps.

svbgit, 2022-10-13:
> Hi, I think it probably needs an amplitude correction factor (divide by windowSum/2) for FFT amplitude. I add in ComplexToMagnitude function and amplitude is consistent with my test signals. Hope this helps.
@JJLynnXiao, can you explain, what is windowSum/2? Or just write your variant of this line of ComplexToMagnitude function: 		`this->_vReal[i] = sqrt(sq(this->_vReal[i]) + sq(this->_vImag[i]));`