kosme/arduinoFFT #36: The calculated peak frequency is half the actual signal frequency.

JuriShalkin, 2019-10-27:
I use the FFT to analyze the sound spectrum on ESP32. Sampling frequency of 40000 Hz, the number of samples is 512. I use the sound signal to calibrate and verify the operation. In doing so, I get the calculated peak frequency half that of the actual signal frequency. The spectrogram looks like a signal with half the frequency.

kosme, 2019-10-27:
What frequency are you expecting to get and which one are you getting?

JuriShalkin, 2019-10-27:
For example, I give a sound signal with a frequency of 8000 Hz, I get about 4000 Hz.

JuriShalkin, 2019-10-30:
My source https://github.com/JuriShalkin/T-Watch-FFT-Waterfall

JuriShalkin, 2019-11-04:
Sorry, this is my mistake. I used the wrong code to read data from the ADC.

closed 2019-11-04 by JuriShalkin