kosme/arduinoFFT #37: Understanding the Bit swap function in Compute

joelror, 2019-11-26:
The compute method runs a small while loop which shifts 'samples' and modifies j to determine when to run Swap function. I am trying to understand the rationale/theory for this bit of logic. It seems to produce a very random pattern?

kosme, 2019-11-27:
It may look random, but it is not. It also took me a while to understand it but if you look at the resulting order of the array, it is arranging the elements in an analog of a [Butterfly diagram](https://en.wikipedia.org/wiki/Butterfly_diagram). The elements that must be computed together end up adjacent to each other. It is quite ingenious and very efficient. I have tried to achieve the same result with other, more obvious to understand, algorithms but this one beats them in terms of processing speed.

joelror, 2019-11-27:
Thanks for the response!

closed 2020-02-21 by kosme