kosme/arduinoFFT #38: Wrong peak frequencies (at sampling rates > ~10000 Hz)

nordblick2, 2020-01-07:
I observed some confusing peak values doing FFT with your lib. After researching a couple of
evenings and debugging without finding a solution, i would like to discuss that here. Maybe I 
did something wrong or misinterpreted something.

Problem: The calculated peaks changes with higher sampling rate (starting from ~10 kHz). 

Up to a sampling rate (SF) of about 10 kHz the returned peak frequency matches my 
signal generators frequencies (537Hz and 2100Hz) more or less excactly. For higher SF 
(10k+), the returned peak is more and more off. That offset increases with an increasing SR (in a linear way). Changing number of samples didn't change anything (seems not relevant). Here are some measured values:

```
		Sig. 1		Sig. 2
SF (Hz)		537 Hz		2100 Hz
-------------------------------------------
 5000		 537		 2100
10000		 537		 2100
15000		 791		 3105
20000		1055		 4141
25000		1318		 5176
30000		1582		 6211
40000		2109		 8281
50000		2657		10352
```

Quite interresting 🧐. At least, there seems to be a pattern. So I could find a good correction factor for sampling rates above 10160Hz:
```	
	Frequency[Real Peak] = Frequency[Peak calculated by FFT] / SF * 10160
```
It's good for a workaround, but no real solution nor explaination for my wrong fft-peaks. 

So... Any idea why this "offset" happens or how could i correct that unexpected behaviour? 

--
Btw#1: Generating vReal (as im your examples sketches) didnt show that error (frequency matches FTT_Peak):

```
double cycles = (((SAMPLES-1) * frequency) / SAMPLING_FREQUENCY);
	for (uint16_t i = 0; i < SAMPLES; i++) {
		vReal[i] = int8_t((AMPLITUDE * (sin((i * (twoPi * cycles)) / SAMPLES))) / 2.0);
		vImag[i] = 0.0;
	}
```

--
The signal generator i used for testing is an phone app (f Generator), that is connected 
via audio jack (Audio-GND to ESP-Gnd, L/R-signal routes via capacitor and a voltage divider
(2x10kOhm) to Pin.A0 of an standard NodeMcu/ESP32. Tested also with a direct connection from head phone jack -> A0) and a microphone. Same offset here...

nordblick2, 2020-01-07:
One additional (and probably important) note: The maximum possible frequency seems to be at 10160/2. For my last tested SF = 50kHz it should be around 25kHz, but is in fact at 5090Hz (signal higher than that will produce a decreasing peak).  Something weird I think ....

kosme, 2020-01-07:
Weird behavior indeed. Will take a look at it.

kosme, 2020-01-09:
I reread the symptoms you described and finally noticed this:
> Btw#1: Generating vReal (as im your examples sketches) didnt show that error (frequency matches FTT_Peak):

That sounds like it is a hardware problem. Most probably, the ADC is not sampling as fast as it is supposed to. I suggest you check that.

nordblick2, 2020-01-09:
Mmmh, any idea how to check that? I already used a different ESP32, without any difference. Probably a problem with the underlying (mostly slower) analogRead()?

nordblick2, 2020-01-09:
Ok, I did a quick test and can (likely) confirm, that my ESP can only reach a sampling rate round around that 10k-value:

```
ulong t0 = micros();
for (int i = 0; i < 1000; i++) {
	analogRead(pin);
}
printf("Frequency: %5.0f Hz\n", ((float)samples * 1000000 / (micros() - t0)));
```
Output:
```
Frequency: 10299.000 Hz
Frequency: 10302.000 Hz
Frequency: 10301.000 Hz
Frequency: 10308.000 Hz
Frequency: 10403.000 Hz
Frequency: 10401.000 Hz
Frequency: 10303.000 Hz
Frequency: 10302.000 Hz
Frequency: 10302.000 Hz
Frequency: 10305.000 Hz
Frequency: 10302.000 Hz
Frequency: 10325.000 Hz
Frequency: 10326.000 Hz
Frequency: 10304.000 Hz
Frequency: 10308.000 Hz
Frequency: 10307.000 Hz
Frequency: 10302.000 Hz
Frequency: 10325.000 Hz
```

Not sure (yet), if I can change that sampling rate. Have to take some look into the spec....

tariqmatech, 2020-01-14:
I am not an expert here, I've been reading for almost a week now for one of my projects and still trying to figure out how things work.

I came across an article that I guess maybe of help to you.

you should give "free-running" a try as analogRead is slow when it comes to multiple sampling. have a look at this article.

[https://blog.yavilevich.com/2016/08/arduino-sound-level-meter-and-spectrum-analyzer/](https://blog.yavilevich.com/2016/08/arduino-sound-level-meter-and-spectrum-analyzer/)

tariqshiwani, 2020-01-14:
I am not an expert here, I've been reading for almost a week now for one of my projects and still trying to figure out how things work.

I came across an article that I guess maybe of help to you.

you should give "free-running" a try as analogRead is slow when it comes to multiple sampling. have a look at this article.

[https://blog.yavilevich.com/2016/08/arduino-sound-level-meter-and-spectrum-analyzer/](https://blog.yavilevich.com/2016/08/arduino-sound-level-meter-and-spectrum-analyzer/)

nordblick2, 2020-01-28:
Already thought about free running mode, but I was using an esp8266 and the arduino-libs seems not to provide an free running mode "out of the box". I got myself some ESP32s now and their sample rate is ~30k out of the box (enough for my project, but  esp32s-platformio-libs have much better hardware access. So I will see how high I could get the sampling rate - and report).
Issue/Help request can be closed for now.

closed 2020-01-28 by nordblick2

groberts22, 2022-12-15:
Just wondering if this issue was ever solved. I am having a very similar problem, where my reported frequencies jump dramatically from their actual values (40% or more) for sampling frequencies greater than 10000KHZ.