kosme/arduinoFFT #40: Submission of ratio table for windowing function

MarScaper, 2020-01-18:
Thanks for your work. I see that ratio table is in your todo list. 

I calculated the table for my own need. Here It is...
FLOAT WindowCompensationFactor[10] =
{
    1*2,                        /* rectangle (Box car) */
    1.8549343278*2, /* hamming */
    1.8554726898*2, /* hann */
    2.0039186079*2, /* triangle (Bartlett) */
    2.8163172034*2, /* nuttall */
    2.367347436*2,  /* blackman */
    2.7557840395*2, /* blackman nuttall */
    2.7929062517*2, /* blackman harris*/
    3.5659039231*2, /* flat top */
    1.5029392863*2  /* welch */
};

Compensation seems to be fine in my project... :)
[image] 

Hope It helps,
Sebastien.

kosme, 2020-01-22:
Thanks. I am still thinking how to integrate it into the code.

MarScaper, 2020-01-23:
In my opinion, the best approach would be to allow compensation within the Windowing() method with a conditional parameter for backward compatibility.

Header could be changed like this...
```
void Windowing(uint8_t windowType, uint8_t dir, bool compensation=false);
``` 

And method implementation like this...
```
void arduinoFFT::Windowing(uint8_t windowType, uint8_t dir)
{// Weighing factors are computed once before multiple use of FFT
// The weighing function is symetric; half the weighs are recorded
	double samplesMinusOne = (double(this->_samples) - 1.0);
	for (uint16_t i = 0; i < (this->_samples >> 1); i++) {
		double indexMinusOne = double(i);
		double ratio = (indexMinusOne / samplesMinusOne);
		double weighingFactor = 1.0;
		// Compute and record weighting factor
		switch (windowType) {
		case FFT_WIN_TYP_RECTANGLE: // rectangle (box car)
			weighingFactor = 1.0;
			break;
		case FFT_WIN_TYP_HAMMING: // hamming
			weighingFactor = 0.54 - (0.46 * cos(twoPi * ratio));
			break;
		case FFT_WIN_TYP_HANN: // hann
			weighingFactor = 0.54 * (1.0 - cos(twoPi * ratio));
			break;
		case FFT_WIN_TYP_TRIANGLE: // triangle (Bartlett)
			weighingFactor = 1.0 - ((2.0 * abs(indexMinusOne - (samplesMinusOne / 2.0))) / samplesMinusOne);
			break;
		case FFT_WIN_TYP_NUTTALL: // nuttall
			weighingFactor = 0.355768 - (0.487396 * (cos(twoPi * ratio))) + (0.144232 * (cos(fourPi * ratio))) - (0.012604 * (cos(sixPi * ratio)));
			break;
		case FFT_WIN_TYP_BLACKMAN: // blackman
			weighingFactor = 0.42323 - (0.49755 * (cos(twoPi * ratio))) + (0.07922 * (cos(fourPi * ratio)));
			break;
		case FFT_WIN_TYP_BLACKMAN_NUTTALL: // blackman nuttall
			weighingFactor = 0.3635819 - (0.4891775 * (cos(twoPi * ratio))) + (0.1365995 * (cos(fourPi * ratio))) - (0.0106411 * (cos(sixPi * ratio)));
			break;
		case FFT_WIN_TYP_BLACKMAN_HARRIS: // blackman harris
			weighingFactor = 0.35875 - (0.48829 * (cos(twoPi * ratio))) + (0.14128 * (cos(fourPi * ratio))) - (0.01168 * (cos(sixPi * ratio)));
			break;
		case FFT_WIN_TYP_FLT_TOP: // flat top
			weighingFactor = 0.2810639 - (0.5208972 * cos(twoPi * ratio)) + (0.1980399 * cos(fourPi * ratio));
			break;
		case FFT_WIN_TYP_WELCH: // welch
			weighingFactor = 1.0 - sq((indexMinusOne - samplesMinusOne / 2.0) / (samplesMinusOne / 2.0));
			break;
		}

		if( compensation ){
			weighingFactor *= WindowCompensationFactor[windowType];
		}

		if (dir == FFT_FORWARD) {
			this->_vReal[i] *= weighingFactor;
			this->_vReal[this->_samples - (i + 1)] *= weighingFactor;
		}
		else {
			this->_vReal[i] /= weighingFactor;
			this->_vReal[this->_samples - (i + 1)] /= weighingFactor;
		}
	}
}
```
Easy to set and no impact on existing code for users.

HorstBaerbel, 2020-04-09:
See the [develop branch](https://github.com/kosme/arduinoFFT/tree/develop) and / or pull request #42 where this is implemented.

kosme, 2023-12-28:
@MarScaper do you agree with the statement on issue #61 ??