kosme/arduinoFFT #45: Change request double type to float

mrguen, 2020-03-08:
Thanks for this great library. I am now using it on STM32 and the double type is 64 bits so it takes a lot of memory when using the actual "double" type that is used everywhere. 

I changed them to float. This is much more convenient and does not visibly reduce  the precision. For usual Arduino boards like the UNO float and double datatypes are equivalent 32 bits data.

HorstBaerbel, 2020-04-09:
See the [develop branch](https://github.com/kosme/arduinoFFT/tree/develop) and / or pull request #42.

mrguen, 2020-07-01:
@HorstBaerbel Thanks. I had the dev branch working well, using a 4096 values float array on my STM32F401RCT6 with Arduino Core STM32

HorstBaerbel, 2020-07-01:
Damn. Should be fixed in pull request #50.

mrguen, 2020-07-03:
@HorstBaerbel 

Thanks for the dev version. I am not testing it on an Atmega1284 and saw these:

1) I think there is a little error in the examples. When creating a synthetic wave
  float cycles = (((samples-1) * signalFrequency) / samplingFrequency); //Number of signal cycles that the sampling will read

should be:  
float cycles = (((samples) * signalFrequency) / samplingFrequency); //Number of signal cycles that the sampling will read

The duration of the sampling is indeed = samples / samplingFrequency either you index them from 0 or 1. 

2) Also I wonder if it is normal to have a peak at "0 H", even after removing the DC component?

HorstBaerbel, 2020-07-03:
I don't know as I'm not the maintainer. I'd suggest you open a new issue when you're sure these things are errors, as they have nothing to do with this one. You can also directly point to source code lines using the web interface: Navigate to the file, click the line number, click the "..." menu and then choose the option you'd like, e.g. "create permalink" or even "new issue".

mrguen, 2020-07-03:
Ok, I will do that.

closed 2024-04-24 by kosme