kosme/arduinoFFT #46: Storage issues with sample sizes over 128

kieferlord, 2020-03-14:
Hello,

Your library works great for sample sizes of 128 and below, but when I try anything above that in order to obtain a better resolution, I have storage issues. The specific error message is:

Global variables use 2278 bytes (111%) of dynamic memory, leaving -230 bytes for local variables. Maximum is 2048 bytes.

I have tried moving the vReal and vImag arrays into the void loop() so they would no longer be considered global variables, but this caused a bunch of other problems (crazy things printing, like backwards question marks, etc.). Do you have any suggestions? Thanks.

kosme, 2020-03-15:
That is a hardware related issue. If you want to store more than 128 samples, you need a device with more RAM available.

HorstBaerbel, 2020-04-09:
You could try switching to float values. See the [develop branch](https://github.com/kosme/arduinoFFT/tree/develop) and / or pull request #42.

closed 2023-12-28 by kosme