kosme/arduinoFFT #48: Major peak vs Magnitude data

dimk1, 2020-05-16:
Hi there,

I'm trying to use the library to sample real-time measurements from a DC current sensor. The piece of code responsible for that is the following:

  

> for (uint16_t i = 0; i < samples; i++)
>   {
>     current_mA = ina219.getCurrent_mA();
>     vReal[i] = uint8_t(current_mA);
>     vImag[i] = 0.0; //Imaginary part must be zeroed in case of looping to avoid wrong calculations and overflows
>   }
>   /* Print the results of the simulated sampling according to time */
>   Serial.println("Data:");
>   PrintVector(vReal, samples, SCL_TIME);
>   FFT.Windowing(vReal, samples, FFT_WIN_TYP_HANN, FFT_FORWARD);	/* Weigh data */
>   Serial.println("Weighed data:");
>   PrintVector(vReal, samples, SCL_TIME);
>   FFT.Compute(vReal, vImag, samples, FFT_FORWARD); /* Compute FFT */
>   Serial.println("Computed Real values:");
>   PrintVector(vReal, samples, SCL_INDEX);
>   // Serial.println("Computed Imaginary values:");
>   // PrintVector(vImag, samples, SCL_INDEX);
>   FFT.ComplexToMagnitude(vReal, vImag, samples); /* Compute magnitudes */
>   Serial.println("Computed magnitudes:");
>   PrintVector(vReal, (samples >> 1), SCL_FREQUENCY);
>   majPeak = FFT.MajorPeak(vReal, samples, samplingFrequency);
>   Serial.print("Major Peak: ");
>   Serial.println(majPeak, 6);

samples are set to 64 and sampling frequency to 2000.
Q1: is sampling frequency measured in Hz in this code? So in my example is it 2kHz?
Q2: is my code correct for sampling the data from the in219 sensor?
Q3: when I print out the results I get the following measurements:

**> [13:07:10:063] Computed magnitudes:**
> [13:07:10:069] 0.000000Hz 3331.2492
> [13:07:10:074] 31.250000Hz 1438.9001
> [13:07:10:079] 62.500000Hz 32.3680
> [13:07:10:079] 93.750000Hz 43.3368
> [13:07:10:085] 125.000000Hz 9.0984
> [13:07:10:091] 156.250000Hz 72.3520
> [13:07:10:091] 187.500000Hz 101.3240
> [13:07:10:096] 218.750000Hz 21.6699
> [13:07:10:101] 250.000000Hz 64.0806
> [13:07:10:101] 281.250000Hz 41.0935
> [13:07:10:107] 312.500000Hz 23.0520
> [13:07:10:112] 343.750000Hz 12.0868
> [13:07:10:112] 375.000000Hz 27.2012
> [13:07:10:118] 406.250000Hz 16.8139
> [13:07:10:124] 437.500000Hz 45.7704
> [13:07:10:124] 468.750000Hz 42.1793
> [13:07:10:129] 500.000000Hz 4.8751
> [13:07:10:135] 531.250000Hz 25.4468
> [13:07:10:135] 562.500000Hz 16.2070
> [13:07:10:141] 593.750000Hz 7.6919
> [13:07:10:146] 625.000000Hz 16.9842
> [13:07:10:146] 656.250000Hz 33.2346
> [13:07:10:151] 687.500000Hz 30.6472
> [13:07:10:157] 718.750000Hz 53.2407
> [13:07:10:163] 750.000000Hz 13.5191
> [13:07:10:163] 781.250000Hz 37.6957
> [13:07:10:169] 812.500000Hz 11.4377
> [13:07:10:173] 843.750000Hz 41.6263
> [13:07:10:173] 875.000000Hz 41.9199
> [13:07:10:179] 906.250000Hz 55.0802
> [13:07:10:185] 937.500000Hz 22.7566
> [13:07:10:185] 968.750000Hz 48.2572

**> [13:07:10:190] Major Peak: 183.070256**

Since this is a DC current sensor (so mostly a steady signal with a few fluctuations), it's logical that the algorithm calculates the biggest magnitudes at 0Hz and 31.25Hz. However, what is the meaning of the Major Peak in that case? I would expect it would print the frequency where the biggest magnitude exists, that is 0Hz. Instead it prints "183.07"  which seems to be close to the frequency of 187.5Hz, which corresponds to the biggest magnitude **excluding** 0Hz and 31.5Hz. So what is the meaning of this value?

DrDiettrich, 2020-05-16:
Am 16.05.2020 um 13:31 schrieb dimk1:

> **> [13:07:10:190] Major Peak: 183.070256**
> 
> Since this is a DC current sensor (so mostly a steady signal with a
> few fluctuations), it's logical that the algorithm calculates the
> biggest magnitudes at 0Hz and 31.25Hz.

The 0Hz Amplitude is not related to a frequency, it's the signal offset.
For a DC current signal this is the mean current, with the other
amplitudes based on that offset.

> However, what is the meaning of the Major Peak in that case? I would
> expect it would print the frequency where the biggest magnitude
> exists, that is 0Hz. Instead it prints "183.07"  which seems to [...]

The real major peak exists near 31.25Hz, it's a bug in the library that 
the first (non-zero) frequency amplitude is always ignored in the major 
peak determination.

The analog peak value is not normally represented in a single digital 
peak value. A frequency between the discrete frequency values is 
distributed to their neighbour values. Paint a smooth curve over the 
values and you'll find a peak higher than the listed digital values. So 
the real frequency and amplitude is computed from an interpolation 
around the most outstanding amplitudes. Again the library is not perfect 
in the interpolation of the true peak frequency and value.

DoDi

dimk1, 2020-05-16:
So, is it correct to ignore the first calculation @0Hz for the analysis and only keep the rest of the values? And how do you mean that the 0Hz amplitude is the mean current? It's value in the example above is 3331 and the current fluctuates around 92mA.

DrDiettrich, 2020-05-17:
Am 16.05.2020 um 23:32 schrieb dimk1:
> So, is it correct to ignore the first calculation @0Hz for the
> analysis and only keep the rest of the values?

Right, the first peak occurs around 31.25 Hz. This special case, with no 
valid value to the left, is not covered by the peak detection function.

> And how do you mean
> that the 0Hz amplitude is the mean current? It's value in the example
> above is 3331 and the current fluctuates around 92mA.

You should know yourself how the measured values are related to the DC 
current - shunt resistance, amplification...

DoDi