kosme/arduinoFFT #51: Usage of multiple FFT's using the same code

srinivas27194, 2020-07-02:
Hello,

I have written a code that reads two signals from two ADC. One signal is captured for 4096 samples and another signal is captured for 512 samples. I am sampling both the signals at 500 Hz. 

I initially compute FFT for the signal with 4096 samples and find out the peak frequency. This works perfectly.

When I compute the FFT for the second signal, it gives me random values.  I have created separate objects for both the FFT's. Kindly let me know if this is possible or should I destruct the first FFT before running the second one. If that's the case, kindly let me know the syntax for the destruct function.

kosme, 2020-07-02:
It is possible to do so. My guess is that you are not resetting the vImag array back to zero before computing the second signal. Line 67 of the example FFT_01 states that not resetting it to zero leads to wrong results.

srinivas27194, 2020-07-03:
[FFT code.docx](https://github.com/kosme/arduinoFFT/files/4869138/FFT.code.docx)

I have used two different sets and also have reset vImag to zero. But still, the second FFT gives random values. I have attached the code for your reference. Kindly look into it and let me know if there is any mistake.

srinivas27194, 2020-07-16:
I have used two different sets and also have reset vImag to zero. But still, the second FFT gives random values. I have attached the code for your reference. Kindly look into it and let me know if there is any mistake.

https://github.com/kosme/arduinoFFT/files/4869138/FFT.code.docx

a0079700, 2020-09-10:
@kosme 
I would like to compute FFTs of two different signals with different samples. I have already tried two different combinations.

Case 1: initialize two different objects outside the Void main() and void loop(), one for 4096 samples FFT and another object for 512 samples FFT

float vReal_temp[4096];
float vReal1[4096];
float vImag1[4096];
float weighingFactors1[4096];

ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal1, vImag1, 4096, 500, weighingFactors);

float vReal2[512];
float vImag2[512];
float weighingFactors2[512];
ArduinoFFT<float> FFT1 = ArduinoFFT<float>(vReal2, vImag2, 512, 500, weighingFactors2);

void loop()
{

microseconds = micros();
  for(int i=0; i<4096; i++)
  {
      vImag1[i] = 0.0;
      vReal1[i] =(1000.0/3.0)*(3.3/(4095.0))*analogRead(32);
      vReal_temp[i]=(1000.0/3.0)*(3.3/(4095.0))*analogRead(33);
      while(micros() - microseconds < sampling_period_us){}
      microseconds += sampling_period_us;
  }

  FFT.windowing(FFTWindow::Hann, FFTDirection::Forward);
  FFT.compute(FFTDirection::Forward);
  FFT.complexToMagnitude();
///some code for peak detection and serial print the value////

  for(int i=0; i<512; i++)
  {
      vImag2[i] = 0.0;
      vReal2[i] =vReal_temp[i];
  }
  FFT1.windowing(FFTWindow::Hann, FFTDirection::Forward);
  FFT1.compute(FFTDirection::Forward);
  FFT1.complexToMagnitude();
///some code for peak detection and serial print the value////
}

In this case, first i compute the fast Fourier transform using the object "FFT". It works well. after that i compute Fourier transform using the second object FFT1. But FFT1 simply prints some random values. Please note that the real and imaginary vectors of two different object FFT an FFT1 are completely different. Also, i am making sure that the imaginary vector is initialized to zero before i compute the FFTs

Case 2: I declare the FFT object every time the loop executes and initialize the object with different vector size

void loop()
{
microseconds = micros();
  for(int i=0; i<4096; i++)
  {
      vImag[i] = 0.0;
      vReal[i] =(1000.0/3.0)*(3.3/(4095.0))*analogRead(32);
      vReal_temp[i]=(1000.0/3.0)*(3.3/(4095.0))*analogRead(33);
      while(micros() - microseconds < sampling_period_us){}
      microseconds += sampling_period_us;
  }
ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal, vImag, 4096, 500, weighingFactors);
  FFT.windowing(FFTWindow::Hann, FFTDirection::Forward);
  FFT.compute(FFTDirection::Forward);
  FFT.complexToMagnitude();
///some code for peak detection and serial print the value////

  for(int i=0; i<512; i++)
  {
      vImag2[i] = 0.0;
      vReal2[i]=vReal_temp[i];
  }
FFT = ArduinoFFT<float>(vReal2, vImag2, 512, 500, weighingFactors2);
  FFT.windowing(FFTWindow::Hann, FFTDirection::Forward);
  FFT.compute(FFTDirection::Forward);
  FFT.complexToMagnitude();
///some code for peak detection and serial print the value////
}

In this scenario also, i am getting some random value for second FFT calculation.

I am not sure why, for first FFT calculation everything is perfect but for subsequent FFT calculation either by separate objects or same object it is printing some random values.

Your advice or suggestion is highly appreciated.