kosme/arduinoFFT #53: DCRemoval() broken in version from Arduino library manager

s-marley, 2020-07-17:
If I install arduinoFFT (v1.5.5) from the Arduino library manager, `DCRemoval()` returns nonsense when I run it on an ESP32. If I download the libarary from here and use that, it works as intended.

chessweb01, 2020-10-05:
That's because for some reason the library manager version contains this:

    for (uint16_t i = 1; i < ((samples >> 1) + 1); i++)

instead of this:

    for (uint16_t i = 0; i < this->_samples; i++)

s-marley, 2020-10-07:
@chessweb01 Thank you for clarifying the problem. How can we get this change pushed to verison used by the Arduino library manager? I have no idea how this process works!

domints, 2023-12-28:
I think it might be fixed now, Arduino downloads latest version 1.6.1 which contains, I think, proper DCRemoval function:
```
void arduinoFFT::DCRemoval() {
  // calculate the mean of vData
  double mean = 0;
  for (uint16_t i = 0; i < this->_samples; i++) {
    mean += this->_vReal[i];
  }
  mean /= this->_samples;
  // Subtract the mean from vData
  for (uint16_t i = 0; i < this->_samples; i++) {
    this->_vReal[i] -= mean;
  }
}
```

Maybe this issue should be closed? ;)  I came from Scott's video from 2020 and was wondering if this issue was fixed or do I need to plug latest github code in my Arduino ;)

kosme, 2023-12-28:
@domints Thank you for bringing this to my attention. There was supposed to be an answer from October 8, 2020, stating that the changes had been included in the then-current version (1.5.6).

closed 2023-12-28 by kosme