kosme/arduinoFFT #54: Detecting wrong major Peak values at low frequencies

mrguen, 2020-08-11:
Hello,

Thanks for the good work. Of course the FFT is an approximation with a couple of error sources. Here I have a problem to detect low fundamental frequencies.

I am sampling a sinusoidal signal from a function generator at Fs = 16 KHz. When the  signal frequency is under approximately 500 Hz, the result of the majorPeak() computation are wrong. The lower limit of the frequency detection depends on the window used. So I tested all the windows it shows

| Windowing |  Lower limit Hz  |  Precision above the limit  |
|---|---|---|
|   |   |   | 
|   |   |   | 
|   |   |   | 

|Without windowing | 
Rectangle
Hamming
Hann
Triangle
Nuttall
Blackman
Blackman Nuttall
Blackman Harris
FLT TOP
Welch

DrDiettrich, 2020-08-11:
Am 11.08.2020 um 11:33 schrieb Thierry GUENNOU:

> Thanks for the good work. Of course the FFT is an approximation with a couple of error sources. Here I have a problem to detect low fundamental frequencies.
> 
> I am sampling a sinusoidal signal from a function generator at Fs = 16 KHz.

Then measure an exact number of waves to satisfy the seamless 
requirement - no windowing required then!

DoDi

mrguen, 2020-08-11:
@DrDiettrich  Do you mean skip to frequency counting instead of FT?

DrDiettrich, 2020-08-11:
Am 11.08.2020 um 17:19 schrieb Thierry GUENNOU:
> @DrDiettrich  Do you mean skip to frequency counting instead of FFT?

Depends on what you want to achieve. You can determine the base 
frequency first, then switch to FFT for detection of harmonics.

DoDi

mrguen, 2020-08-11:
Yes of course but we are talking about the FFT library here. I wonder if 1) there might be a bug 2) we can do something to measure the low frequency component. 3) In consequence I also realize that it is not possible to use this to compute a full spectrum, unless the frequencies under 600 Hz don't mind which is very doubtfull for anyone looking for audio spectral analysis. I have no idea if this can be better or not.

mrguen, 2020-08-12:
@DrDiettrich 

Sorry, there was nothing wrong with the detection of MajorPeak. I was in fact sampling @ 600 KHz so there were not enough periods in the samples to compute the fequencies

closed 2020-08-12 by mrguen