kosme/arduinoFFT #62: Alternate Current 50 Hz frequency spectrum

terrasmart, 2021-01-08:
Hello all, this is not an issue but a need for clarification.

I'm working on a project where I would like to use this library to get a frequency spectrum of a 50 Hz alternate current by a current transducer 0-5V connected to an Arduino UNO through the ADC. 

I'm specially looking at the FFT_03 example. 

SinceI want to investigate the spectrum of the first 2-3 seconds of an induction motor start-up, this is the 'period' or interval I would to sample the ADC signal.

I would like to understand the meaning of _samplingFrequency_ and _sampling_period_us_.

Nyquist's theorem states that a periodic signal must be sampled at more than twice the highest frequency component of the signal. This means that, in case of an AC of 50 Hz, the  _samplingFrequency_ should be at least 100 Hz.  

According to the code of FFT_03, the _sampling_period_us_ would be 10 secs. If the samplingFrequency is 200 Hz, the sampling period would be 5 secs.

Question is: can I simply adjust the _samplingFrequency_ till I get the wanted _sampling_period_us_ ? Is there any other correlation that I must consider ?

What is the range of the "Computed magnitudes" given by the library ?

Thanks anybody for the help.

Michele

d4vidyo, 2021-01-22:
Hi,
samplingFrequency is the value which states how often per second the ADC is reading the Voltage.
sampling_period_us is just how much time is between each of those reads in microseconds or 10^-6 seconds.

In the FFT_03 example there is another value right above samplingFrequency called "samples". This value tells your ADC how many readings it should take before the Arduino performs the FFT. Because in order to calculate the FFT you need a complete array of readings.

So if you have a samplingFrequency of 100Hz this would give you a sampling_period_us value of 10 000 microseconds or 0.01 seconds. Which means every 0.01 seconds your ADC is measuring the Voltage once. This happens for samples amount of times so if you set your samples value to 64 you will end up with an array of 64 Voltage readings which would take 0.64 seconds to measure. 

So to end up with 2-3seconds of sampling time you just need to adjust the samples value. Be sure it is a power of 2 (2, 4, 8, 16, ...1024 etc.). If you set it to 256 samples this would take 2.56 seconds which would be right in your desired time.

However if you want to see if the motor has some weird frequency's going on that are above 50Hz you would need to increase the samplingFrequency but i think you understood that part about Nyquist.

Hope this helps!
David

kosme, 2023-12-28:
I am closing this thread. Feel free to reopen it if required.

closed 2023-12-28 by kosme