kosme/arduinoFFT #63: formula for windowing algorithms

jerabaul29, 2021-01-13:
I am confused about the formula used for the windowing algorithms. Do you have a reference document describing these?

For example:

https://github.com/kosme/arduinoFFT/blob/566803e9ca2b6ff50895da2791fcfbf80f8fd23a/src/arduinoFFT.cpp#L305-L307

does not seem to agree with the amplitude or energy correction factors here:

https://community.sw.siemens.com/s/article/window-correction-factors

Maybe I am just confused though.

kosme, 2021-01-13:
I don't have any reference documents; as it says on the README this is a
fork from an abandoned project.

The correction factors on the Siemens paper apply to the value after the
FFT, so the value is in Frequency domain. The windowing factors apply to
the signal before the FFT, so the value is in Time domain.

jerabaul29, 2021-01-14:
Mmh, if I remember well FFT(a f) = a FFT(f), so applying the correction before or after the transform should be the same, right? Or am I missing something? ...

hugke729, 2021-01-14:
@jerabaul29 : your expression is correct if `a` is a constant, but in this case it is a window. Look at the second panel in the first figure of the article you linked to. Note, also, that `weighingFactor` is a confusing name here. It's not a single factor, it's a window.. Finally, per issue #61, the Hann window isn't quite right.

jerabaul29, 2021-01-14:
Mmh, agree I should have been clearer and write that:

FFT(a * signal * windowing) = a * FFT(signal * windowing),

I wrote in the way I did because it is the usual notation when working with proofs of linearity on vector spaces.

But with a the constant scaling factor, it should be the same before or after applying the FFT I would think? ...

Just to be clear: I am aware that the Hanning window looks like a 'bell', I am thinking of the factor 0.54 that seems to disagree with other sources I have found.

DrDiettrich, 2021-01-15:
On 14.01.21 22:18, JR wrote:
> Mmh, agree I should have been clearer and write that:
>
> FFT(a * signal * windowing) = a * FFT(signal * windowing),
>
> I wrote in the way I did because it is the usual notation when working
> with proofs [...]

Fourier transformation assumes an infinitely repetitive signal, with the 
sample spanning an integral number of waves of the base (lowest) 
considered frequency. Thus windowing must assure steady continuation 
from the end of a sample to its start.

If sample end and start do not match exactly in all derivatives then the 
transformation contains artefacts caused by the jumps from the sample 
end to the start level. Choice of the most appropriate windowing 
(smoothing) function allows to minimize these artefacts.

DoDi

jerabaul29, 2021-01-15:
Yes, I am aware of that. What we discuss here is the factor to use so that the windowing does not change the energy content.

jerabaul29, 2021-02-20:
If someone else looks into this and finds this thread. I took a bit of time to double check this. I am pretty confident that these formula are not right:

https://github.com/kosme/arduinoFFT/blob/bb99065f9aad9e5781c5bf9e77613bdd33f0074d/src/arduinoFFT.cpp#L244-L249

I take as a reference the reference numpy documentation, which I have much trust in:

https://numpy.org/doc/stable/reference/generated/numpy.hanning.html

https://numpy.org/doc/stable/reference/generated/numpy.hamming.html

And a table from univ. Zurich for the "scaling coefficients":

https://www.physik.uzh.ch/local/teaching/SPI301/LV-2015-Help/lvanlsconcepts.chm/Scaling_Smoothing_Windows.html

which are in agreement with the previous link.

I am personally moving to kissfft, as it fits my needs better.

jerabaul29, 2021-02-20:
Sorry, the Hamming should be good, but the Hann is probably wrong.

jerabaul29, 2021-02-20:
(but I still think that these windows are not compensated; which is fine, but may be worth having a line of comment about the need to compensate to keep either amplitude or energy constant :) ).