kosme/arduinoFFT #74: Implement C++ template

blazoncek, 2022-07-01:
Sometimes double precision is not necessary and may just make computations slow.

It would be nice if the library was made using template so that both double and float could be used.

If this is not possible (due to compatibility with C) then a float version would also work in addition to double.

blaz-r, 2022-07-01:
This is implemented on develop branch. It also includes some other optimizations.

It'd be nice to have that on main, but it's probably not due to compatibility.

blazoncek, 2022-07-02:
Thanks for pointing that out.
Still there are non-float constants and functions used there. Changing those may further optimise the execution and storage used.

kosme, 2023-05-26:
@blazoncek Do you consider that the current code on the main branch accomplishes what you are pointing out??

blazoncek, 2023-05-26:
Hi. Thanks for the update but I fail to see templates in main branch, though I can see template implementation in the develop branch.

I have one more remark. There are quite a few floating point constants used in the code. Unless those are suffixed by `f` those are stored as double precision regardless of the type used in template. I would recommend to suffix at least those that do not benefit of the precision to reduce the code size.

blazoncek, 2024-03-07:
Seems like this can be closed with v2.
Thank you for the effort.

closed 2024-03-07 by blazoncek