kosme/arduinoFFT #84: Performance++

core-c, 2023-04-11:
In the sourcecode i see divisions being done inside a loop.
Here is one example:
`    for (uint16_t i = 0; i < this->_samples; i++) {
      this->_vReal[i] /= this->_samples;
      this->_vImag[i] /= this->_samples;
    }
`
A multiplication is usually much faster executed than a division.
You could calculate the reciprocal once, and then use that value inside the loop to multiplicate with.
It will also eliminate the typecasts done when a division is performed (_samples is an integer, but _vReal[] & _vImag[] are doubles), because that typecast is then executed once outside the loop: double rpSamples = 1.0 / this->_samples;

/ 2.0 == * 0.5            But not in speed of execution

BjornTheProgrammer, 2023-04-24:
@core-c, does the compiler not already optimize this loop? If you have a way to verify that there is a statistically significant speed up, I could make a pull request.

core-c, 2023-04-25:
I can check that out, and get back on it.

There are more things that can be optimized though: Like assigning a value inside a loop, from a line of code that always produces the same result..  after some minor calculation.
Such things seem fine enough on a fast processor, maybe, but not so on an AVR, i think. Those MCUs run at low clockspeeds, where every tick counts.

BjornTheProgrammer, 2023-04-25:
So I compiled the code changing this block of code
```c++
for (uint16_t i = 0; i < this->_samples; i++) {
    this->_vReal[i] /= this->_samples;
    this->_vImag[i] /= this->_samples;
}
```
to
```c++
for (uint16_t i = 0; i < this->_samples; i++) {
    double reciprocal = 1 / this->_samples;
    this->_vReal[i] *= reciprocal;
    this->_vImag[i] *= reciprocal;
}
```
The compiled code for each did change. I ran it through a disassembler, but the resulting output was completely different. It turns out that Arduino compiles with the gcc `-Os` flag enabled. This means that Arduino optimizes for space instead of speed. The compiler still might also be optimizing this loop, but there is no way to know, so there might actually be a speed benefit for doing this. It warrants additional testing/research.

core-c, 2023-04-26:
-Os optimizes for size.
There are other options you can try, like the most common: -O1, -O2, -O3 (increasingly more speed optimization).

In any case, you should calculate that reciprocal ONCE outside the for-loop.
Now it calculates _reciprocal_ **this->_samples** times, while it keeps resulting the same value (_samples value does not change).

I will see if i can make things faster, in C.
But i am really thinking of writing it in AVR assembler (because i need it for my own projects). But for that, first the algorithm/calculation must be efficient.

FintasticMan, 2023-04-30:
The develop branch has an `FFT_SPEED_OVER_PRECISION` macro that you can define that does what you're proposing.

kosme, 2023-05-26:
@core-c I am intrigued about checking out the URLs you mention

core-c, 2023-07-08:
http://wiki.openmusiclabs.com/wiki/ArduinoFHT
https://github.com/imagest108/arduino/blob/master/libraries/PlainFFT/PlainFFT.cpp
https://github.com/imagest108/arduino/tree/master/libraries
https://github.com/Evg33/ArduinoFHT
https://github.com/kosme/arduinoFFT

..

core-c, 2023-07-08:
sorry for a very late reply.. i was busy with RL.   kosme

kosme, 2023-07-25:
Both mentioned optimizations are implemented on version 1.6.1

closed 2023-07-25 by kosme