kosme/arduinoFFT #92: First two bins 0 Hz and 62.5 Hz always an order of magnitude larger than all other bins

systembolaget, 2024-03-26:
Great library, I very much appreciate the work you have done!

Adafruit Metro Mini. Adafruit MAX4466 electret Microphone and Amplifier. Testing with sine waves from smartphone app audio generator. I can simply ignore those two bins, but I'm curious as to why that is so. Here, a 1000 Hz sine wave, 0 Hz and 62.5 Hz bins excluded.

[image]

SlipperyTrout, 2024-04-10:
This isnt really an issue for the library so youre kinda askin in the wrong place. But 60Hz is right around the frequency of mains electricity so that could be a liklely culprit. For the 0Hz spike im pretty sure its an artefact of how FFT works and the maths behind it.

kosme, 2024-04-10:
@systembolaget From your graph, I can infer that you are using 128 samples at an 8000 Hz sampling rate. So, the frequency of the first bin is always zero, while the frequency of the second bin depends on your specific sampling rate.

I am intrigued by the fact that you say that they are an order of magnitude larger than ALL other bins. Does that include the main frequency of your signal (1000 Hz)??? If that is the case, there is definitely a very noisy source in your circuit. At first glance, I would side with @SlipperyTrout regarding his diagnostic.

kosme, 2024-04-17:
After further analysis, the problem DOES exist. It is an earlier manifestation of what is reported in issue #93.

addarecords, 2024-05-30:
I have this issue as well. My Second bin is massive. Any ideas on a workaround for this?

ex.
Computed magnitudes:
0.000000Hz 866.9319
9.765625Hz 443705.3405
19.531250Hz 1259.4838
29.296875Hz 1043.9390
39.062500Hz 869.9087
48.828125Hz 855.4211
58.593750Hz 1516.0719
68.359375Hz 830.0902
78.125000Hz 604.5652
87.890625Hz 1415.7364
97.656250Hz 656.0529
107.421875Hz 881.4611
117.187500Hz 892.0497
126.953125Hz 972.8943
136.718750Hz 702.1117
146.484375Hz 858.1995
156.250000Hz 891.8153
166.015625Hz 1025.3826
175.781250Hz 678.8556
185.546875Hz 953.5659
195.312500Hz 1023.4730
205.078125Hz 799.7329
214.843750Hz 778.0143
224.609375Hz 1129.0133

kosme, 2024-06-06:
@addarecords This could be a problem with your sampling function. Check the solution on issue #99 and confirm that you are not committing the same mistake.