kosme/arduinoFFT #99: Cannot detect frequency from sound sensor in v2

deltakiro, 2024-05-08:
> ArduinoFFT: v2.0.2 (Does not give the expected output)
> ArduinoFFT: v1.6.2 (Give the expected output)
> ArduinoIDE: 2.3.2
> Board: Arduino UNO R3
> Sound sensor: LM393 (its blue in color with 3 pins)

So I'm tryna make an Arduino based guitar tuner for my school project and I've noticed that the frequency for any of the strings is not detected by the board but if I run the code from v1.6.2 it just works perfectly.

I'll explain the problem with the HIGH E string (the result is the same for all of the strings).

So the frequency for the HIGH E string is suppose to be around 330Hz but I just get the wrong value(i.e. 7.74) for the code in v2 and the frequency for other strings are also wrong. I tried checking it with a tuner and it just says my guitar is in tune and also v1.6.2 agrees with the tuner. I'm still new to the Arduino area please let me know if this is a problem on my end or from the library which needs to be fixed.

### Code form v1.6.2
```cpp
#include "arduinoFFT.h"

#define samples 128
#define samplingFrequency 1000
 
arduinoFFT FFT = arduinoFFT();
 
unsigned int samplingPeriod;
unsigned long microseconds;

double vReal[samples];
double vImag[samples];

void setup() {
  Serial.begin(115200);
  samplingPeriod = round(1000000*(1.0/samplingFrequency));
}

void loop() {
    for(int i=0; i<samples; i++)
    {
        microseconds = micros();  
        vReal[i] = analogRead(1);
        vImag[i] = 0;
        while(micros() < (microseconds + samplingPeriod));
    }
 
    FFT.Windowing(vReal, samples, FFT_WIN_TYP_HAMMING, FFT_FORWARD);
    FFT.Compute(vReal, vImag, samples, FFT_FORWARD);
    FFT.ComplexToMagnitude(vReal, vImag, samples);
    double peak = FFT.MajorPeak(vReal, samples, samplingFrequency);
 
  Serial.println(peak);
  peak = peak * 0.990;
  return peak;
}
```

### Code from v2:
```cpp
#include  <arduinoFFT.h>

#define samples 128
#define samplingFreq 1000
double vReal[samples];
double vImag[samples];

ArduinoFFT<double> FFT = ArduinoFFT<double>(vReal, vImag, samples, samplingFreq);
 
int samplingPeriod;
unsigned long microSeconds;
 
void setup() {
  Serial.begin(115200);
  samplingPeriod = round(1000000*(1/samplingFreq));
}

void loop() {  
  frequency();
  delay(100);
}

double frequency() {
 for(int i = 0; i < samples; i++) {
    microSeconds = micros(); 
    vReal[i] = analogRead(5);
    vImag[i] = 0;
    while(micros() < (microSeconds + samplingPeriod));
  }
  FFT.windowing(vReal, samples, FFTWindow::Hamming, FFTDirection::Forward);
  FFT.compute(vReal,  vImag, samples, FFTDirection::Forward);
  FFT.complexToMagnitude(vReal, vImag, samples);

  double peak = FFT.majorPeakParabola(vReal,  samples, samplingFreq);
  peak = peak * 0.99;
  Serial.println(peak);
}
```
Frequency from v1.6.2:

[[image]
](url)
Frequency from v2

[image]

kosme, 2024-05-09:
The code you run for v1.6.2 and v2.0.x has very subtle differences. One of those subtle differences is the cause of your problem. Specifically, the problem is in the sampling period definition. 

Making changes so that variables have a common name between versions, 
In the v2.0.x code, you define `samplingPeriod = round(1000000*(1/samplingFreq));` 
In the v1.6.2 code, you define `samplingPeriod = round(1000000*(1.0/samplingFreq));`
In both cases, you use `#define samplingFreq 1000`
For v2.0.x, the resulting `samplingPeriod` is 0; for v1.6.2, `samplingPeriod` equals 1000.

Your options are:
1. `#define samplingFreq 1000.0`
2. `samplingPeriod = round(1000000*(1.0/samplingFreq));`
3. Make `samplingFreq` either a `float` or `double` variable.

closed 2024-05-09 by kosme

deltakiro, 2024-05-12:
Ohh thank you it works now!!!