kosme/arduinoFFT #102: Forward followed by Reverse transform introduces constant offset into original data

jremington, 2024-10-29:
This simple test code generates and transforms #include "arduinoFFT.h" in the forward and reverse directions. The resulting sine wave is offset by a constant -3.89.  Please check!

`#include "arduinoFFT.h"

const uint16_t samples = 64; //This value MUST ALWAYS be a power of 2
const float signalFrequency = 500;
const float samplingFrequency = 5000;
const uint8_t amplitude = 100;

/*
These are the input and output vectors
Input vectors receive computed results from FFT
*/
float vReal[samples];
float vImag[samples];

/* Create FFT object */
ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal, vImag, samples, samplingFrequency);

#define SCL_INDEX 0x00
#define SCL_TIME 0x01
#define SCL_FREQUENCY 0x02
#define SCL_PLOT 0x03

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Ready");
  /* Build raw data */
  float ratio = twoPi * signalFrequency / samplingFrequency; // Fraction of a complete cycle stored at each sample (in radians)
  for (uint16_t i = 0; i < samples; i++)
  {
    vReal[i] = amplitude * sin(i * ratio);/* Build data with positive and negative values*/
     vImag[i] = 0.0; //Imaginary part must be zeroed in case of looping to avoid wrong calculations and overflows
  }
  /* Print the results of the simulated sampling according to time */
  Serial.println("Data:");
  PrintVector(vReal, samples, SCL_TIME);
  FFT.compute(FFTDirection::Forward); /* Compute FFT */
  FFT.compute(FFTDirection::Reverse);
  Serial.println("data, after forward/reverse:");
  for (uint16_t i = 0; i < samples; i++)
  {
    vReal[i] += 3.89;  //add back the mysterious offset as a test.
  }
  PrintVector(vReal, samples, SCL_FREQUENCY);
}

void loop(){}

void PrintVector(float *vData, uint16_t bufferSize, uint8_t scaleType)
{
  for (uint16_t i = 0; i < bufferSize; i++)
  {
    float abscissa;
    /* Print abscissa value */
    switch (scaleType)
    {
      case SCL_INDEX:
        abscissa = (i * 1.0);
	break;
      case SCL_TIME:
        abscissa = ((i * 1.0) / samplingFrequency);
	break;
      case SCL_FREQUENCY:
        abscissa = ((i * 1.0 * samplingFrequency) / samples);
	break;
    }
//    Serial.print(abscissa, 6);
//    if(scaleType==SCL_FREQUENCY)
//      Serial.print("Hz");
//    Serial.print(" ");
    Serial.println(vData[i]);
  }
  Serial.println();
}`

jremington, 2024-10-29:
Another test, with the data scaled up by a factor of 10, shows that the introduced offset is also scaled up by a factor of 10. But it affects only the real part of the result. The imaginary result is correct (~ 0.0).

I checked the forward FFT calculation using Octave, and those results agree completely. There is clearly something wrong with the inverse calculation.

Cleaned up test code displays the full results of both the forward and reverse calculation.

```
#include "arduinoFFT.h"

const uint16_t samples = 64; //This value MUST ALWAYS be a power of 2
const float signalFrequency = 500;
const float samplingFrequency = 5000;
const float amplitude = 1000;

/*
These are the input and output vectors
Input vectors receive computed results from FFT
*/
float vReal[samples];
float vImag[samples];

/* Create FFT object */
ArduinoFFT<float> FFT = ArduinoFFT<float>(vReal, vImag, samples, samplingFrequency);

#define SCL_INDEX 0x00
#define SCL_TIME 0x01
#define SCL_FREQUENCY 0x02
#define SCL_PLOT 0x03

void setup()
{
  Serial.begin(115200);
  while(!Serial);
  Serial.println("Ready");
  /* Build raw data */
  float ratio = twoPi * signalFrequency / samplingFrequency; // Fraction of a complete cycle stored at each sample (in radians)
  for (uint16_t i = 0; i < samples; i++)
  {
    vReal[i] = amplitude * sin(i * ratio);/* Build data with positive and negative values*/
    vImag[i] = 0.0; //Imaginary part 
  }
  /* Print the results of the simulated sampling according to time */
  Serial.println("Data:");
  PrintVector(vReal, samples, SCL_TIME);
  FFT.compute(FFTDirection::Forward); /* Compute FFT */

  Serial.println("forward transform:");
    for (uint16_t i = 0; i < samples; i++)
  {
    Serial.print(vReal[i]);
    Serial.print(", ");
    Serial.println(vImag[i]);
  }
  Serial.println();
FFT.compute(FFTDirection::Reverse);
  Serial.println("inverse transform:");
    for (uint16_t i = 0; i < samples; i++)
  {
    Serial.print(vReal[i]);
    Serial.print(", ");
    Serial.println(vImag[i]);
  }
  Serial.println();
}

void loop(){}

void PrintVector(float *vData, uint16_t bufferSize, uint8_t scaleType)
{
  for (uint16_t i = 0; i < bufferSize; i++)
  {
    Serial.println(vData[i]);
  }
  Serial.println();
}
```

kosme, 2024-10-30:
@jremington This should be fixed on the new release. Reopen the issue if necessary.

closed 2024-10-30 by kosme