kosme/arduinoFFT #109: arduino FFt not found

Jasmanbid, 2024-12-30:
I'm constantly facing with this error
C:\Users\Hamidreza\AppData\Local\Temp\.arduinoIDE-unsaved20241131-12796-4ux5d7.w7ped\sketch_dec31a\sketch_dec31a.ino:11:1: error: 'arduinoFFT' does not name a type; did you mean 'ArduinoFFT'?
   11 | arduinoFFT FFT = arduinoFFT(vReal, vImag, SAMPLES, SAMPLING_FREQUENCY);
      | ^~~~~~~~~~
      | ArduinoFFT
C:\Users\Hamidreza\AppData\Local\Temp\.arduinoIDE-unsaved20241131-12796-4ux5d7.w7ped\sketch_dec31a\sketch_dec31a.ino: In function 'void analyzeSound()':
C:\Users\Hamidreza\AppData\Local\Temp\.arduinoIDE-unsaved20241131-12796-4ux5d7.w7ped\sketch_dec31a\sketch_dec31a.ino:56:5: error: 'FFT' was not declared in this scope
   56 |     FFT.windowing(vReal, SAMPLES, FFT_WIN_TYP_HAMMING, FFT_FORWARD);
      |     ^~~

Compilation error: 'arduinoFFT' does not name a type; did you mean 'ArduinoFFT'?

kosme, 2024-12-31:
The error message is giving you the solution. You are using arduinoFFT, with a lowercase **a** at the start. The type is called ArduinoFFT, with an uppercase **A** at the start. Use **ArduinoFFT**, not **arduinoFFT**.

closed 2024-12-31 by kosme

Jasmanbid, 2025-01-02:
But when I change it it gives :Compilation error: class template argument deduction failed

Jasmanbid, 2025-01-02:
My main problem is with: ArduinoFFT FFT(vReal, vImag, SAMPLES, SAMPLING_FREQUENCY);

kosme, 2025-01-03:
You are missing the type when creating the ArduinoFFT object. Check the documentation here:

[https://github.com/kosme/arduinoFFT/wiki/api#constructors](https://github.com/kosme/arduinoFFT/wiki/api#constructors)